PX4/flight_review #234: Jinja2 import is borken

eduardkieser, 2022-04-05:
The jinja2 verion is not pinned in the requirements.txt file, mine installed 3.1.1, which breaks the Markup import in templates.py
```
from jinja2.utils import markupsafe 
Markup = markupsafe.Markup()
```
Seems to do the trick.

eduardkieser, 2022-04-06:
It seems to be a combination between the Bokeh and Jinja version.

luvwinnie, 2022-04-07:
I'm facing the same problem, it seems like by upgrading the bokeh==2.1.1 to bokeh==2.4.2(latest) can be fix this error.

luvwinnie, 2022-04-07:
I have tested that if upgrade from bokeh 2.1.1 to 2.4.2 would broke the program, instead make sure that Jinja2 use for example 3.0.0 instead of latest 3.1.1.

bkueng, 2022-06-14:
Should be resolved with https://github.com/PX4/flight_review/pull/238

closed 2022-06-14 by bkueng