PX4/flight_review #243: serve.py --show giving parser error

UASOP, 2022-07-27:
Hey there, 

Just cloned the latest main/master and when running the ./serve.py --show command I am getting this error. All requirement modules were installed per the install instructions. 

Traceback (most recent call last):
  File "./serve.py", line 22, in <module>
    from tornado_handlers.download import DownloadHandler
  File "/home/user/flight_review/app/tornado_handlers/download.py", line 18, in <module>
    from helper import get_log_filename, validate_log_id, \
  File "/home/user/flight_review/app/plot_app/helper.py", line 28, in <module>
    from libevents_parse.parser import Parser
ModuleNotFoundError: No module named 'libevents_parse'

Did not see this issue with previous versions of the flight review. Tried finding this libevents_parse it is talking about and unsure if I am looking in the right places.

bkueng, 2022-07-27:
Hi

You need to initialize the submodule with
```
git submodule init
git submodule update
```

UASOP, 2022-07-31:
That did the trick! Thank you good sir.

closed 2022-07-31 by UASOP