PX4/flight_review #255: serve.py -f   error

mengchaoheng, 2023-02-24:
```Console 
(base) ➜  app git:(main) ✗ ./serve.py -f /Users/mch/Documents/FlightLog/08_39_27.ulg 
GET[log]=/Users/mch/Documents/FlightLog/08_39_27.ulg
DB access failed: <class 'sqlite3.OperationalError'> no such table: Logs
Downloading https://px4-travis.s3.amazonaws.com/Firmware/master/_general/airframes.xml
Download error: [Errno 2] No such file or directory: '/Users/mch/Proj/flight_review/app/plot_app/../../data/cache/airframes.xml.552616d5-b01a-4c6d-9346-ed0e75e7cacb'
<class 'IndexError'> (vehicle_local_position_setpoint): list index out of range
<class 'IndexError'> (vehicle_local_position_groundtruth): list index out of range
<class 'IndexError'> (vehicle_attitude_groundtruth): list index out of range
<class 'IndexError'> (vehicle_angular_velocity_groundtruth): list index out of range
<class 'IndexError'> (vehicle_attitude_groundtruth): list index out of range
<class 'IndexError'> (vehicle_angular_velocity_groundtruth): list index out of range
<class 'IndexError'> (vehicle_attitude_groundtruth): list index out of range
<class 'IndexError'> (vehicle_angular_velocity_groundtruth): list index out of range
<class 'IndexError'> (vehicle_local_position_setpoint): list index out of range
<class 'IndexError'> (vehicle_local_position_setpoint): list index out of range
<class 'IndexError'> (vehicle_local_position_setpoint): list index out of range
<class 'IndexError'> (vehicle_local_position_setpoint): list index out of range
<class 'IndexError'> (tecs_status): list index out of range
<class 'IndexError'> (actuator_controls_1): list index out of range
<class 'IndexError'> (actuator_controls_1): list index out of range
<class 'IndexError'> (actuator_outputs): list index out of range
<class 'IndexError'> (actuator_outputs): list index out of range
<class 'IndexError'> (vehicle_imu_status): list index out of range
<class 'IndexError'> (vehicle_imu_status): list index out of range
/Users/mch/opt/anaconda3/lib/python3.9/site-packages/bokeh/models/plots.py:815: UserWarning: 
You are attempting to set `plot.legend.click_policy` on a plot that has zero legends added, this will have no effect.

Before legend properties can be set, you must add a Legend explicitly, or call a glyph method with a legend parameter set.

  warnings.warn(_LEGEND_EMPTY_WARNING % attr)
<class 'IndexError'> (fifo data): list index out of range
<class 'IndexError'> (fifo data): list index out of range
<class 'IndexError'> (distance_sensor): list index out of range
<class 'IndexError'> (airspeed): list index out of range
<class 'KeyError'> (vehicle_status): 'failsafe_and_user_took_over'
<class 'IndexError'> (failsafe_flags): list index out of range
Downloading https://px4-travis.s3.amazonaws.com/Firmware/master/_general/parameters.xml
Download error: [Errno 2] No such file or directory: '/Users/mch/Proj/flight_review/app/plot_app/../../data/cache/parameters.xml.59744743-bfc5-4924-8f69-cfcf69a68873'
Downloading https://px4-travis.s3.amazonaws.com/Firmware/master/_general/all_events.json.xz
Download error: [Errno 2] No such file or directory: '/Users/mch/Proj/flight_review/app/plot_app/../../data/cache/events.json.xz.b8b0d915-b6f9-4a83-811f-69ca0b30b678'
```

bkueng, 2023-02-28:
Did you run `setup_db.py`?

mengchaoheng, 2023-02-28:
Yes, I have been run it! @bkueng 

```Console 

parallels@ubuntu-linux-20-04-desktop: /Users/mch/Proj/flight_review/app$ ./setup_db.py 
creating log directory /Users/mch/Proj/flight_review/app/plot_app/../../data/log_files
creating cache directory /Users/mch/Proj/flight_review/app/plot_app/../../data/cache
creating kml directory /Users/mch/Proj/flight_review/app/plot_app/../../data/cache/kml
creating overview image directory /Users/mch/Proj/flight_review/app/plot_app/../../data/cache/img
creating DB at /Users/mch/Proj/flight_review/app/plot_app/../../data/logs.sqlite
parallels@ubuntu-linux-20-04-desktop: /Users/mch/Proj/flight_review/app$ ./serve.py -f /Users/mch/Documents/FlightLog/log/2023-02-24/07_42_12.ulg         
GET[log]=/Users/mch/Documents/FlightLog/log/2023-02-24/07_42_12.ulg
Downloading https://px4-travis.s3.amazonaws.com/Firmware/master/_general/airframes.xml
<class 'IndexError'> (vehicle_local_position_groundtruth): list index out of range
<class 'IndexError'> (position_setpoint_triplet): list index out of range
<class 'IndexError'> (vehicle_attitude_groundtruth): list index out of range
<class 'IndexError'> (vehicle_angular_velocity_groundtruth): list index out of range
<class 'IndexError'> (vehicle_attitude_groundtruth): list index out of range
<class 'IndexError'> (vehicle_angular_velocity_groundtruth): list index out of range
<class 'IndexError'> (vehicle_attitude_groundtruth): list index out of range
<class 'IndexError'> (vehicle_angular_velocity_groundtruth): list index out of range
<class 'IndexError'> (tecs_status): list index out of range
<class 'IndexError'> (actuator_controls_1): list index out of range
<class 'IndexError'> (actuator_controls_1): list index out of range
<class 'IndexError'> (actuator_outputs): list index out of range
<class 'IndexError'> (actuator_outputs): list index out of range
<class 'IndexError'> (vehicle_imu_status): list index out of range
<class 'IndexError'> (vehicle_imu_status): list index out of range
/Users/mch/opt/anaconda3/lib/python3.9/site-packages/bokeh/models/plots.py:815: UserWarning: 
You are attempting to set `plot.legend.click_policy` on a plot that has zero legends added, this will have no effect.

Before legend properties can be set, you must add a Legend explicitly, or call a glyph method with a legend parameter set.

  warnings.warn(_LEGEND_EMPTY_WARNING % attr)
<class 'IndexError'> (fifo data): list index out of range
<class 'IndexError'> (fifo data): list index out of range
<class 'IndexError'> (distance_sensor): list index out of range
<class 'IndexError'> (airspeed): list index out of range
<class 'KeyError'> (vehicle_status): 'failsafe_and_user_took_over'
<class 'IndexError'> (failsafe_flags): list index out of range
Downloading https://px4-travis.s3.amazonaws.com/Firmware/master/_general/parameters.xml
Downloading https://px4-travis.s3.amazonaws.com/Firmware/master/_general/all_events.json.xz
WARNING:tornado.access:404 GET /favicon.ico (::1) 0.62ms

```

closed 2023-06-19 by bkueng