PX4/flight_review #256: Running Flight Review Locally not returning to the script that called it.

Nomed73, 2023-03-14:
I am running flight review locally, from a python script that uses os.system() to run it.  There seems to be no issue in running it. My web browser shows FlightReview, the data and graphs all seem to be correct. However I get the following errors in the console:

```
Waiting for drone to connect...
-- Connected to drone!
Flight Review initiated...
GET[log]=/home/nm/Downloads/FlightLogs/log-1-2023-03-09-17-00-29.ulg
DB access failed: <class 'sqlite3.OperationalError'> no such table: Logs
Downloading https://px4-travis.s3.amazonaws.com/Firmware/master/_general/airframes.xml
Download error: [Errno 2] No such file or directory: '/home/nm/dev/flight_review/app/plot_app/../../data/cache/airframes.xml.31dfdbd5-c553-4852-b282-c087e73fc5de'
<class 'IndexError'> (vehicle_angular_velocity_groundtruth): list index out of range
<class 'IndexError'> (vehicle_angular_velocity_groundtruth): list index out of range
<class 'IndexError'> (vehicle_angular_velocity_groundtruth): list index out of range
<class 'IndexError'> (tecs_status): list index out of range
<class 'KeyError'> (manual_control_setpoint): 'mode_slot'
<class 'IndexError'> (vehicle_angular_acceleration): list index out of range
<class 'IndexError'> (vehicle_torque_setpoint): list index out of range
<class 'IndexError'> (vehicle_thrust_setpoint): list index out of range
<class 'IndexError'> (actuator_servos): list index out of range
<class 'IndexError'> (vehicle_imu_status): list index out of range
<class 'IndexError'> (vehicle_imu_status): list index out of range
<class 'IndexError'> (vehicle_imu_status): list index out of range
/home/nm/dev/FlightLogs/venv/lib/python3.8/site-packages/bokeh/models/plots.py:815: UserWarning: 
You are attempting to set `plot.legend.click_policy` on a plot that has zero legends added, this will have no effect.

Before legend properties can be set, you must add a Legend explicitly, or call a glyph method with a legend parameter set.

  warnings.warn(_LEGEND_EMPTY_WARNING % attr)
<class 'IndexError'> (vehicle_angular_acceleration): list index out of range
<class 'IndexError'> (fifo data): list index out of range
<class 'IndexError'> (fifo data): list index out of range
<class 'IndexError'> (distance_sensor): list index out of range
<class 'IndexError'> (system_power): list index out of range
<class 'IndexError'> (airspeed): list index out of range
Downloading https://px4-travis.s3.amazonaws.com/Firmware/master/_general/parameters.xml
Download error: [Errno 2] No such file or directory: '/home/nm/dev/flight_review/app/plot_app/../../data/cache/parameters.xml.3e39702f-053e-4af6-8dc1-eac1458d693a'
Failed to get event parser: [Errno 2] No such file or directory: '/home/nm/dev/flight_review/app/plot_app/../../data/cache/metadata'
```
Although I get this information in the console, Flight Review does open up in Firefox and shows the data and graphs correctly. However, 
I believe this could be keeping the script to continue running and I cannot get back to other tasks in the script. 

What should I be looking into to have this running without issues and return to the script.

bkueng, 2023-03-15:
The error won't change the behavior though. The server must be kept running as long as you want to use it from the browser (zooming into a graph will dynamically reload data). If you want the script to continue, make sure to run it in the background.

Does the directory `/home/nm/dev/flight_review/app/plot_app/../../data/cache/` exist?

Nomed73, 2023-03-15:
@bkueng I tried looking for that directory. I just don't know where to look in for the /../../ folders. I did look through most and have not seen them. Suppose it doesn't exist, where should I create it or is it supposed to be created on its own? 

I have tried running it in the background but I haven't found the way to make it work properly. I will keep searching for a solution and post here if I find one.

bkueng, 2023-03-16:
It's `/home/nm/dev/flight_review/data/cache/`.

If you use python, you can do:
```
import subprocess
subprocess.Popen(["serve.py","-f","file.ulg"])
```

bkueng, 2023-06-19:
I'm assuming this is resolved. Let me know if not.