PX4/flight_review #259: IsADirectoryError When Running Production

brennan-toofon, 2023-05-23:
Hello,

I am getting many `IsADirectoryError: [Errno 21] Is a directory: '/opt/service/plot_app/../config_user.ini'` in my docker logs when I build the site using `./init-letsencrypt.sh`.

The app works locally and even when hosted using `docker-compose up`, so why would there be errors now? Has anyone come across this?

When accessing the site in the browser, I get the blank "500 Internal Server Error / nginx/1.23.4" page after logging in.

Thanks

bkueng, 2023-05-24:
Hi
I'm not using the container, so I have not come across this. Let me know if you find anything.

brennan-toofon, 2023-07-06:
Deployment is successful if I don't read `config_user.ini` by commenting out lines 16 and 17 in `plot_app/config.py`. I am also not sure what "add app/config_user.ini with respective stages." means in the readme.

bkueng, 2024-12-16:
Hi did you ever resolve this? The user config is just a file to insert your customizations, like a map key. This is why it's outside of docker.