PX4/flight_review #267: Why Step Response for (Roll/Pitch/Yaw) Rate is Zero?

MuhammadBilal1, 2023-11-02:
### Describe the bug

Hi,
    I am using PX4 ([1.14.0](https://github.com/PX4/PX4-Autopilot/releases/tag/v1.14.0)) in my drone which is using [Pixhawk 6C](https://holybro.com/products/pixhawk-6c), but since last few days, I am getting an issue:
The Step Response for (Roll/Pitch/Yaw) Rate is **Zero** in the flight logs (PID Analysis).
I need this doing the PID Tuning of my drone.

**1-** Sometimes "Step Response for Roll Rate" at ZERO line
**2-** Sometimes "Step Response for Pitch Rate" is at ZERO line
**3-** Sometimes "Step Response for Yaw Rate" is at ZERO line
**4-** Sometimes "Step Response for Roll/Pitch Rate" is at ZERO line

Why is this happening? I am doing manual PID Tuning, flight behavior seems fine (not 100%, but not shaking or crashing), but response is zero.
I am attaching some of the logs here:

### To Reproduce

Performing manual tuning.
Almost all of the flights are done in **Position** flight mode.
I am using **UM982** dual antenna GPS.

### Expected behavior

The PID Analysis graphs should show the correct values of the PID for each rate instead of a flat line at 0.

### Screenshot / Media

[image]

[image]

### Flight Log

**Roll & Pitch** are at 0
https://review.px4.io/plot_app?plots=pid_analysis&log=60815230-a1fb-40fd-a64b-54abfcb1d143

**Roll, Pitch & Yaw** at 0
https://review.px4.io/plot_app?plots=pid_analysis&log=08c14c20-38ca-404e-aee4-30c531839f56

**Pitch** at 0
https://review.px4.io/plot_app?plots=pid_analysis&log=f804b7d3-5bf6-49cc-bef6-3473c7e4d8b4

**Roll & Pitch** are at 0
https://review.px4.io/plot_app?plots=pid_analysis&log=edcf0b09-5ea0-4dce-a5d5-115349aea314

**Roll, Pitch & Yaw** are available
https://review.px4.io/plot_app?plots=pid_analysis&log=4e796e69-30c9-4199-9063-81c6259ba9f6

### Software Version

PX4 ([1.14.0](https://github.com/PX4/PX4-Autopilot/releases/tag/v1.14.0))

### Flight controller

Pixhawk 6C

### Vehicle type

Multicopter

bkueng, 2023-11-06:
Hi
This is when there's not enough motion. The result is best if you fly in Stabilized/Altitude with fast and full stick changes.

From the log it looks like the `MC_YAWRATE_P` is a bit too high causing noise on the outputs. I would also set `MC_YAWRATE_D` to 0.

MuhammadBilal1, 2023-11-09:
> Hi This is when there's not enough motion. The result is best if you fly in Stabilized/Altitude with fast and full stick changes.
> 
> From the log it looks like the `MC_YAWRATE_P` is a bit too high causing noise on the outputs. I would also set `MC_YAWRATE_D` to 0.

Hi,
    Thank you for your response.
Ok. Is this possible to do the PID Tuning by looking at the following graphs?
* Roll Angle
* Roll Angular Rate
* Pitch Angle
* Pitch Angular Rate
* Yaw Angle
* Yaw Angular Rate

A person told me that we can do it from the above graphs like seeing the difference between the **Setpoint** & the **Estimated** lines.

**For Example:** 
- If the difference between Estimated and Setpoint is greater, then we need to increase the **I** value to increase the speed of response.
- If the Estimate isn't reaching the Setpoint, then we should increase the value of **P**.
- If the Estimate is over the Setpoint, then we should decrease the value of **P** for that angle.

bkueng, 2023-11-14:
Yes you can use the graphs.
Have a look at the documentation for that:
https://docs.px4.io/main/en/config_mc/pid_tuning_guide_multicopter_basic.html
https://docs.px4.io/main/en/config_mc/pid_tuning_guide_multicopter.html

closed 2023-11-15 by MuhammadBilal1