PX4/flight_review #296: The entire service is single-threaded, so when uploading files, no other operations can be performed.

ThreeDogsAl, 2024-12-05:
The entire service is single-threaded, so when uploading files, no other operations can be performed.What should i do

ThreeDogsAl, 2024-12-05:
[image]
I directly ran the server.py file. Then when uploading a file, as shown in the screenshot, if I tried to access the /upload page while the file was being uploaded, it would keep loading. Only after entering the /plot_app?log=6c1d2f41-a923-4e48-a02b-f55963457781 page would the /upload page in the other window finally load.

ThreeDogsAl, 2024-12-05:
I've noticed the --num-procs parameter, but I'm using Windows.

ThreeDogsAl, 2024-12-05:
> That is actually not correct. Requests are handled asynchronously. Even with a single thread, the tornado web server serves requests during an upload. In addition, the `serve.py` script allows to pass `--num-procs` to run multiple processes.

I've noticed the --num-procs parameter, but I'm using Windows.

ThreeDogsAl, 2024-12-05:
[image]
When I'm uploading a file, if I make a request to `/upload` at the same time, the `print("=================")` statement won't be executed at all.

ThreeDogsAl, 2024-12-06:
@bkueng What should i do

ThreeDogsAl, 2024-12-06:
I'm deploying it on Windows and directly starting the project with the script "server.py". There is the problem mentioned above. During the uploading process, I can't perform other operations, such as viewing the "/browse" interface or the "/upload" interface. Are there any additional configurations? 

gs
***@***.***

bkueng, 2024-12-09:
Sorry, but I cannot help you here, especially not with Windows. You might want to look into Tornado, and how it works on Windows.

ThreeDogsAl, 2024-12-10:
@bkueng thanks. I am encountering the same issue on Linux: after uploading the .ulg file, other requests get blocked during the process of generating the plot.

bkueng, 2024-12-11:
The plot generation might take some time depending on your cpu and log size. You can enable timing information here: https://github.com/PX4/flight_review/blob/main/app/config_default.ini#L29

bkueng, 2024-12-16:
Closing, you're welcome to improve plot loading time if you find anything.

closed 2024-12-16 by bkueng