PX4/flight_review #297: download rate too slow

EhsanZibaei, 2024-12-05:
Hi everyone,
I’m working on a research project focused on diagnosing UAV failures automatically using statistical methods. To support this, I need to download a large number of flight logs from [review.px4.io](http://review.px4.io/). However, the server seems to be limiting the download speed to 20 Kb/s.
[image]

Is this a common issue? Are there any alternative methods or tools to speed up the download process?

Thanks in advance for your help!

bkueng, 2024-12-09:
That is indeed awfully slow. Download is generally not super fast, but in my experience at least 10x from what you see. Did you try on multiple occasions?
Are you using the download script from the repo?

EhsanZibaei, 2024-12-09:
I tried again today, and this time it was faster. around 3 Mb/s. Maybe it was a server issue

bkueng, 2024-12-16:
Ok good to know, thanks. There's not much I can do here though.

closed 2024-12-16 by bkueng