PX4/flight_review #299: Flight Review 3D Visualization: Yaw Angle Incorrectly Rotated by 90°

amarwingxpand, 2024-12-18:
I noticed while reviewing older logs that Flight Review’s 3D visualization appears to be incorrectly interpreting the aircraft’s yaw angle. It is rotated by about 90 degrees, causing the fixed-wing model to appear perpendicular to the intended takeoff direction, as illustrated in the attached image.
[image]

bkueng, 2024-12-19:
Can you provide a log?

z77z, 2025-01-06:
I also found that whether it was the log imported a long time ago or the log imported now, it was the same, and the course was deflected 90 degrees.

bkueng, 2025-01-06:
Fixing it here: https://github.com/PX4/flight_review/pull/301

z77z, 2025-01-07:
航向正常显示了，但是俯仰和滚转反了
https://logs.px4.io/3d?log=519b3946-76b4-4e03-b6cf-90a1c0e1198f

bkueng, 2025-01-08:
Sorry I didn't check roll/pitch. It should be correct this time: https://github.com/PX4/flight_review/pull/303

[image]