PX4/flight_review #308: Http 500 error

dlech, 2025-03-31:
I'm getting a 500 error code when trying to view https://logs.px4.io/3d?log=b0705488-ffa7-4e24-a6a8-1707131b277d.

Anyone else get the same? Any idea why?

bkueng, 2025-04-10:
The problem is that the log contains no valid utc time ('time_utc_usec' is 0). Which simulator is this? Ideally we change this in PX4.

I'm also adding a work-around: https://github.com/PX4/flight_review/pull/309

dlech, 2025-04-10:
It was from the (not classic) gazebo simulator. (`make px4_sitl gz_x500_baylands`). I don't remember if this log was from the main branch or the v1.15.4 release tag, but it was one of those two.

bkueng, 2025-04-11:
It's this one then: https://github.com/PX4/PX4-Autopilot/blob/main/src/modules/simulation/gz_bridge/GZBridge.cpp#L625

@StefanoColli do you want to look into that?

closed 2025-04-11 by bkueng